sinara-hw/BaseMod #2: BaseMod remove power detector in next revision?

gkasprow, 2018-08-18:
_From @jbqubit on December 4, 2017 21:26_

- [ ] Consider removing power detector from Allaki

Context:
- Usefulness of power detector is unclear:
  - It only connects to 24-bit ADC, which is very slow and software/gateware support not funded yet.
  - Gain flatness of Sayma/BaseMod should be fairly good compared with the amplifiers, filters etc that are used after it. Thus one generally wants to do power-levelling at the point of load, not at Sayma RF connector
  - BaseMod will have a low-noise MSPS ADC with programmable-gain front-end. Users can place a small power detector near their load. e.g. ADL5511. This will allow closed-loop feedback to the RF at the point where the user cares. Moreover, thermal environment of external power detector can be better controlled, allowing improved stability. External power detector can potentially also be used for using feedback to remove thermal transients in amplifiers etc.
- Power detector is the biggest component on BaseMod. BaseMod is very cramped, removing this will help quite a bit. It also reduces the amount of software support needed for BaseMod (and the number of things that need production testing).

Risk:
- Low, just remove a component!

---

Moving from https://github.com/m-labs/sinara/issues/269. 

@hartytp asks
> Does anyone object to removing the power detector from BaseMod v1.1? cf #397

@dhslichter responds
> The point of the power detector is to noise-eat at the output of the DACs. What is it costing us?

@hartytp responds

Yes, but only when the switch is off, and only via the 24-bit ADC.

Main cost is space: as @sbouhabib pointed out elsewhere, this is the largest component on BaseMod and fitting it on was actually quite challenging. If we remove it then we can make the layout a bit less tight.

Not a huge deal, but it's not totally cost free so unless you really plan to use it then we should scrap it.
-----------------

My comments.... 

Agreed that space occupied by the power detector design is huge. Especially because it involves impedance controlled traces. Design complexity due to the inclusion of the detector can only be described as very high. 

[image]

It also consists of 27 BOM items. AD8363 is about $7/ea and HMC349 switch is about $5. Component cost is not high. 

> I don't think it's actually that useful because:
> It is read out by the 24-bit ADC which is very slow

The original intent to compensate for long-term drift. Not for fast feedback. As you point out the power can only be monitored when the RF output to the physics target is off (using HMC349 switch). This was never intended for fast feedback.

Agreed that this is something that probably should have been culled from the original design. 

_Copied from original issue: sinara-hw/sinara#411_

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:27_

So, are we happy removing it then? Is that a decision or a discussion?

gkasprow, 2018-08-18:
_From @jbqubit on December 4, 2017 21:29_

@hartytp No. This is not a decision. This is moving the discussion to a separate thread for discussion.

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:31_

optimistic ;)

gkasprow, 2018-08-18:
_From @jbqubit on December 4, 2017 21:31_

Let's add to errata list #269 only once a decision is made.

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:32_

ack

gkasprow, 2018-08-18:
_From @dhslichter on December 4, 2017 21:34_

I don't love it enough to fight for it.

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:36_

> I don't love it enough to fight for it.

IME, that's generally a good indication that it's not useful enough to be worth keeping. I'll keep the pistols ready in case you change your mind though ;)

gkasprow, 2018-08-18:
_From @dhslichter on December 4, 2017 21:38_

I think my rationale is that probably I will be rolling my own special-purpose AFEs for the applications we want to be ultra-precise on and so we can always go back to this if we want to use it for design ideas later.

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:40_

Sounds good. Let's get a stripped down basic AFE that works really nicely. People can then add their own pet features to it as they desire.

gkasprow, 2018-08-18:
_From @dtcallcock on December 4, 2017 21:41_

The two large red areas @jbqubit highlights on the left of the top image are the rf switches.  These would presumably be retained even if the detector was removed.

I like the power detector because I spend a lot of time in the lab plugging in the rf power meter to debug DDSs or calibrate out changes in output power vs frequency.  The latter is probably not that important for driving AOMs but will make the board much nicer for broadband applications.

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:43_

But, won't you still end up doing that even with the power detector, since you can only calibrate up to the switch? The gain flatness of this board should be really quite good, so the amplifiers/filters you put after it will ultimately limit the gain flatness as much as the stuff up to the switch.

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:44_

Anyway, you have an analog input, so your best bet AFAICT is to just add a small power detector + pickoff near your load. Then everything is calibrated. Plus, if you do it that way you get to use the nice 5MSPS ADC on the mezzanine so you can do it in-loop...

Edit: we have some thumb nail sized ADL5511 boards that are good to a few GHz, which are perfect for this kind of thing.

gkasprow, 2018-08-18:
_From @dtcallcock on December 4, 2017 21:54_

> But, won't you still end up doing that even with the power detector, since you can only calibrate up to the switch? 

I think having signal sources that output the programmed power regardless of frequency is generally considered a useful thing regardless of the fact that amps aren't perfect.

> The gain flatness of this board should be really quite good.

I imagine the DAC output power is the main issue, especially as you start climbing up towards f_clk/2.

> Anyway, you have an analog input, so your best bet AFAICT is to just add a small power detector + pickoff near your load. 

Given this is an option I don't think it's a big real to remove it.  I just like the option of doing this without adding over a dozen little minicircuits boxes and cables to the Sayma card...

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 21:56_

Well, I think we've heard all possible arguments from each side. Time for a decision. How do we make those  in this project again (do I need to get my CAPSLOCK key out after all)?

gkasprow, 2018-08-18:
_From @hartytp on December 4, 2017 22:56_

I've added the "discussion" label to all Sayma and BaseMod issues that might be contentious. For most of these, the various possibilities have been mapped out, as well as the various arguments for and against each. So, now we just need an executive decision to be made. @jbqubit I assume that you will do that? If so, when you get a chance can you review them, make a decision, summarize the action points at the top of the issue, and remove the discussion label? Feel free to delegate some of this work if you'd rather.

gkasprow, 2018-12-15:
We will need some extra space to implement Sampler AFE + ADC, moreover we do not have any ADCs on Sayma.
I can try to fit all stuff and use remaining 4 channels of LTC ADC for power detector readout.

gkasprow, 2018-12-15:
We have much more space than in rev1. I can fit all we need easily

gkasprow, 2018-12-16:
we will make it
[image]
[image]

gkasprow, 2018-12-16:
@hartytp The AD8363 has power output and temperature output .
We can connect only PD outputs to four free channels of 8-channel LTC2320. It need dedicated drivers.
We can also use 4-channel LTC2324 and additional 8 channel sigma-delta ADC like was used on Sayma.

gkasprow, 2018-12-16:
@hartytp @jbqubit Here is _very_ initial version of board schematics and placement.
Still many things to do, but please have a look at it.
[BaseMod_FMC.PDF](https://github.com/sinara-hw/BaseMod/files/2683179/BaseMod_FMC.PDF)
Missing power budget, not verified logic levels, missing power sequencing analysis, PD outputs not assigned to ADC due to issue above.

hartytp, 2018-12-17:
I'd still really prefer to scrap the power detector since, for the reasons I've outlined before, I really don't think it's very useful: the power v frequency flatness of the DAC + AFE should be pretty good. RF from Sayma is almost never directly applied to an ion. Instead, it always goes via an amplifier and often an AOM. Those other components tend to limit the output power flatness, so there is not much point doing an open-loop calibration on the AFE mezzanine. Instead, it's much better to use the ADCs to do closed-loop feedback (which is what they're there for). Beyond that, if you just want to sanity check the RF levels you're applying to an amplifier then you can use Booster, which has good diagnostics.

IIRC @jbqubit has already made an executive decision to remove these power detectors from this design revision.

> We can connect only PD outputs to four free channels of 8-channel LTC2320. It need dedicated drivers.

No, let's not do that. I don't want to waste board space and money on drivers for these power detectors. 

I also do not want to use the LTC2320, but rather the LTC2324-16, as it's faster. I also do not want to mix up these power detectors with the AFE analog inputs.

hartytp, 2018-12-17:
> @hartytp @jbqubit Here is very initial version of board schematics and placement.

Thanks @gkasprow! Still some work to do, but it looks really good as a starting point.

Shall we finalize the RTM first and then come back to this? At the very least, I'd like to know how much AMC <-> RTM IO we have before choosing ADCs etc

gkasprow, 2018-12-17:
OK, if we remove the power detector, routing of DAC signals would be far easier

hartytp, 2018-12-17:
> OK, if we remove the power detector, routing of DAC signals would be far easier

Good. Let's prioritise the core functionality of this design and make sure we leave enough room for a good, low cross-talk layout.

gkasprow, 2018-12-17:
OK, removed.

closed 2018-12-17 by gkasprow