sinara-hw/BaseMod #5: Sayma AFE cost

gkasprow, 2018-08-18:
_From @hartytp on November 29, 2017 18:36_

- [ ] Review factors that contribute to the cost of Sayma AFE mezzanines and see if we can adjust the design to minimise costs.

- Current BaseMod AFE design costs 500EUR per AFE/2k EUR per Sayma! That's very significant (price will be a key factor in determining user adoption of Sayma compared with competing systems).
-  For Sayma v2.0 I'd really like to get to the point where separating the AFE onto a separate mezzanine does not contribute significantly to the overall cost of Sayma, compared with putting the AFE directly on the Sayma PCB.

Thoughts for cost reduction:
  - BaseMod can be FR408 (or even FR4?) rather than fancy Rodgers material (RF trace widths will need a tweak). Only top-layer needs to be good RF substrate, rest can be FR4 to cut costs.
- Swapping mezzanines from 2 DAC channels each to 4 channels each will help reduce mezzanine cost (and improve mechanics etc as well)
- Using multi-pin Samtec RF connectors will make a huge difference on cost of Sayma. Currently Sayma has 

_Copied from original issue: sinara-hw/sinara#387_

gkasprow, 2018-08-18:
It is roughly 500EUR @ 25pcs.
The most expensive are SMA (TELEGARTNER) and SMP connectors.
Rogers material and process is also quite expensive. We can also use FR408, but here @sbouhabib opinion is needed.

gkasprow, 2018-08-18:
@hartytp regulator price is negligible.

gkasprow, 2018-08-18:
_From @hartytp on November 29, 2017 19:15_

Okay, that's a pretty high cost given how cheap the important components are.

More reason to replace the SMAs + SMPs with something cheaper and better mechanically.

I'd guess FR408 is absolutely fine for the relatively low frequencies we're working at (the trace length isn't that long IIRC). That should also reduce the lead time quite a bit, which is nice!

gkasprow, 2018-08-18:
_From @dhslichter on November 29, 2017 20:54_

For Allaki, I think FR408 is acceptable in terms of loss and impedance-matching quality because the frequencies involved are not aiming to be too high.  One reason we went for Rogers originally was that I had done a bunch of work with it already in testing trace widths etc., and we thought we'd try to put microwaves out directly with mix-mode (recall one can nominally generate 43Ca+ microwave frequencies using mix-mode...).  Let's save the fancy materials for another board.

gkasprow, 2018-08-18:
_From @jbqubit on November 30, 2017 21:21_

Cheaper, easier to manufacture sound like pluses. Specification of a thicker PCB would help with insertion force.

gkasprow, 2018-08-18:
_From @hartytp on November 30, 2017 21:25_

What's the insertion force of one of those 10-pin RF connectors? My guess is it's low enough that we don't need a thicker PCB (which is generally more expensive to manufacture, particularly when using nice laminates like FR408). But, Greg's test PCB will tell us this...

gkasprow, 2018-08-18:
_From @dhslichter on November 30, 2017 21:38_

You can probably make a top layer of FR408 with thicker layers of regular FR4 below (if we only care about the DAC side having good high-frequency loss properties).

gkasprow, 2018-08-18:
@dhslichter Yes, one can also combine FR4 with Rogers.

gkasprow, 2018-08-18:
_From @dhslichter on November 30, 2017 22:52_

Is that not what is being currently done for Allaki?

gkasprow, 2018-08-18:
_From @dhslichter on November 30, 2017 22:53_

Our NIST microwave boards are single layer Rogers on 3 layers of FR4 for cost purposes.

jbqubit, 2019-05-28:
@gkasprow Can probably close this.

closed 2019-05-28 by hartytp