sinara-hw/BaseMod #6: BaseMod rev1 errata

gkasprow, 2018-08-18:
_From @gkasprow on September 4, 2017 19:55_

- [ ] replace AD8363ACPZ-WP  with AD8363ACPZ-R7CT-ND 
- [ ] replace ADT7420UCPZ-R2 with ADT7420UCPZ-RL7TR-ND
- [ ] This board has a lot of pads for potential matching components. At the frequency this AFE is designed to operate at, these are unnecessary. Please remove them from the schematic + layout to simplify things and save space (the boards are quite cramped atm).
- [ ] We do not need an input path that bypasses the InAmp. Please remove these solder jumpers.
- [ ] Zero-index everything (silk, schematic, panel)
- [ ] Make sure that we're using the cheapest acceptable substrates. e.g. FR408 on the RF layer, FR4 everywhere else.
- [ ] Should we add inductors on the +-12V to filter them (these are generated by SMPSs on Sayma and do not have LDOs on BaseMod so may contain high-frequency noise).
- [ ] Add version number to schematic
- [ ] Text is upside down for C4, R40, R42, C63 (although, these are being dropped from V2.0)
- [ ] should probably have a power budget on the schematic like the ones for the EEMs.
- [ ] Input/output connector caps should be 2*1nF on each connector. (Why are there three caps on the input SMA?)
- [ ] add current limiting resistor for attenuator output

_Copied from original issue: sinara-hw/sinara#269_

gkasprow, 2018-08-18:
_From @sbourdeauducq on September 5, 2017 3:19_

As explained in detail several times already, it would be better to have one issue for each item instead of one big errata, and use milestones and labels.

gkasprow, 2018-08-18:
_From @jbqubit on September 22, 2017 17:55_

@gkasprow asked in email
>To do tests of Allaki, I'd need sine wave at the DAC output. The sooner, the better.
What about clocking DAC from 100MHz source?

gkasprow, 2018-08-18:
_From @jbqubit on November 16, 2017 15:40_

- Allaki as-shipped to UMD do not have serial numbers. Please add serial numbers so end-users, WUT and Technosystem can track.

- back-side silkscreen double printing 
[image]

- As-shipped to UMD insulating front-panel missing. @gkasprow is sending the right panel. Photo from @gkasprow below so everybody can see what this will look like. 
[image]

marmeladapk, 2019-01-23:
> replace ADT7420UCPZ-R2 with ADT7420UCPZ-RL7TR-ND

Not changed in current schematics.

> This board has a lot of pads for potential matching components. At the frequency this AFE is designed to operate at, these are unnecessary. Please remove them from the schematic + layout to simplify things and save space

@hartytp @jbqubit Is this still relevant?

> FR408 on the RF layer, FR4 everywhere else.

FR4 is on all layers.

> inductors on the +-12V

Placed on RTM.

> Input/output connector caps should be 2*1nF on each connector

Now there are 2*10nF on inputs, + additional 10pF on output.

> add current limiting resistor for attenuator output

@hartytp @jbqubit Is this still smth to do, or are 0 Ohm resistors on output fine for this purpose?

closed 2019-04-16 by gkasprow