sinara-hw/BaseMod #8: choice of ADC

hartytp, 2018-12-17:
Let's use LTC2324-16 for the BaseMod/MixMod AFE ADC.

If we have enough IO, I'd like to connect 6 diff pairs to the AMC FPGA: SDOA-SDOD, CLKIN, CLKOUT.

We could in principle save IO on the AMC<->RTM connector by:
- Operate in DDR mode and only use 2 data lanes (SDOA, SDOC)
- Put CLKIN generator on the RTM FPGA

hartytp, 2018-12-18:
It looks like we can spare 7 LVDS links between the AMC FPAG and each AFE mezzanine. If that's true, then we should consider using the [LTC2325-16](https://www.analog.com/en/products/ltc2325-16.html#product-overview) as our ADC. I believe this is pin compatible with the LTC2324-16 and a similar cost (this needs checking) but is capable of faster operation. The only downside is that it doesn't support the 2-lane operation that the LTC2324-16 does, but that's not an issue if we have IO to spare.

jbqubit, 2018-12-19:
LTC2325-16 looks good
- 4 channel 
- 5 Msps per channel throughput 
- one cycle latency
- 82 dB SNR at 2 MHz

@gkasprow Is there IO to spare for the LTC2325-16? If nobody has any objections let's lock this in @hartytp.

jbqubit, 2018-12-19:
What I mean is locking in RTM dependencies so @gkasprow has certainty about RTM schematic modifications.

gkasprow, 2018-12-19:
Yes, we have plenty of space.

jbqubit, 2018-12-19:
OK. Update https://github.com/sinara-hw/sinara/issues/605 and let's close this Issue unless something else comes up. :)

marmeladapk, 2019-01-23:
@gkasprow Currently in schematics there is LTC2324. @hartytp Should it be changed according to this thread?

gkasprow, 2019-01-23:
oh, it's missing -16 in the PN. I will fix it.

hartytp, 2019-03-26:
While I'd generally prefer the LTC2325-16 (5MSPS) it's availability seems pretty poor. So, unless someone knows of a reliable source of it, let's go for LTC2324-16 instead.

closed 2019-04-16 by gkasprow