sinara-hw/BaseMod #14: DAC output -> fast comparator -> RTM FPGA input

sbourdeauducq, 2019-03-11:
Is it realistic to add such a path on Sayma RTM?

* Useful as a built-in self-test to check that DAC outputs are toggling *and* synchronized to the FPGA.
* Useful as a back-up synchronization path if we can't get everything to respond properly to SYSREF (AMC FPGA would reorder samples based on what the actual DAC output is).
* One path for each DAC.
* AMC FPGA generates test pattern on JESD204 which is timed by DDMTD core on RTM FPGA.
* RF switch on AFE can be used to disable output during BIST.
* The fast comparator could potentially be a hacked LVDS input buffer in the FPGA, with proper signal conditioning.
* Maximum DAC frequency is 1.2GHz. Maximum data rate received correctly by the FPGA is rated at 1250MBps (625MHz). We can either add a RF switch to avoid overloading the FPGA input, or just hope that it won't cause any ill effects. Based on past experience with Sayma, I prefer the former option.
* If we use an external comparator, the RF switch action can be effected by driving the other comparator input out of DAC range.

gkasprow, 2019-04-16:
@hartytp @sbourdeauducq 
We don't have any RTM LVDS lines anymore. But we still have 5 AMC LVDS pairs.
Shall I connect the comparator directly to the DAC output or after the attenuator or even power amplifier vis some attenuator?
Which comparator to use?
I use in other designs LMH7220 / ADCMP605 / ADCMP604 / LTC6754.
I will add a power switch to disable the comparators when not needed. Some of them have the disable input already.

sbourdeauducq, 2019-04-17:
>  We don't have any RTM LVDS lines anymore. But we still have 5 AMC LVDS pairs.

What are the various LVDS lines currently used for? Ultrascale tends to be a pain for timing-critical cores.

gkasprow, 2019-04-17:
they are used for LEDs and ADC gain control. I can replace them with US LVDS easily.
Here is a schematic of the comparator circuit.
[image]

sbourdeauducq, 2019-04-17:
Yes, maybe we can leave the LEDs to the Ultrascale...
What's the idea here exactly with some control signals going to the RTM FPGA and others to the AMC? How did you decide which lines go to which FPGA?

gkasprow, 2019-04-17:
we have:
- 12 LVDS AMC lines. they will go mostly to the ADC and LEDs
- 8 LVDS RTM lines. They will go to comparators and ADC gain setting
- 15 LVCMOS RTM lines - they are used for attenuator and RF switch control

gkasprow, 2019-05-20:
no comments so closing.

closed 2019-05-20 by gkasprow