sinara-hw/BaseMod #24: R36 power handling

hartytp, 2019-03-26:
@gkasprow in this version, we're still using the higher current pre-amplifiers (compared to the ones on Urukul).

The pre-amps on the current design then we need to watch the power handling on R36 (the amplifier bias resistor). At 65mA this dissipated 425mW. Can it handle that?

In the long run, it might be wise to consider either switching to the same lower current pre-amps we're using on Urukul or using a current source to bias the amps. (If I have time then I may test the later out using an eval board when Sayma v2.0 arrives)

hartytp, 2019-03-26:
Other than that, I don't see any issues with the current design.

gkasprow, 2019-04-16:
replaced with 1218  1W resistor

closed 2019-04-16 by gkasprow