sinara-hw/BaseMod #27: panel design

gkasprow, 2019-05-20:
how should LEDs be marked on the panel?
Note that there is very little space for any marking.
One LED is green, the other is red, so maybe we don't need any marking :)
[image]

hartytp, 2019-05-20:
I’m fine with unlabelled leds if there is so little space. Something generic like “error” is unhelpful anyway

jordens, 2019-05-20:
Nothing for the LEDs. But in and out for the connectors and channel numbers is crucial.

jbqubit, 2020-01-17:
@marmeladapk What does panel design look like as produced?

jbqubit, 2020-01-17:
https://github.com/sinara-hw/Sayma_AMC/issues/121#issuecomment-575673522

Please post photo of what they look like with connectors installed. Helps us all be on the same page about what this looks like before they are ordered in larger number.