sinara-hw/BaseMod #30: Channel zero has lower amplitude

marmeladapk, 2020-01-15:
[image]

Yellow: SMA output from channel 0
Blue: SMA output from channel 1
Violet: Probe of channel 0
Green: Probe of channel 1

Channels were probed on R95, so before amplifier and after filter and balun. It seems that CH0 has higher amplitude initially, but then it gets distorted.

It's consistent on all basemods that we tried.

marmeladapk, 2020-01-15:
[image]

Measuring on DAC output, negative lines.

gkasprow, 2020-01-15:
Try it with lower frequency. Try to swap the scope probes between channels.

marmeladapk, 2020-01-15:
I'll try on sine generator from @sbourdeauducq, but I'll need to swap channels in JCDG.

marmeladapk, 2020-01-20:
Output from modified testmod (yellow, blue - dac ch0; pink, green - dac ch2), averaging from 32 triggers
[image]

I removed baluns and connected SMA directly to +- lines.

Differential signal Ch0:
[image]

Differential signal Ch2:
[image]

Waveforms compared (red: DAC channel 0 differential, white: DAC channel 2 differential)
[image]

So there's a slight difference, but nothing as drastic as we saw on Basemod. Those small differences could be due to solder joints, slightly longer cables which were used to connect pads etc. I'd say that the problem is on Basemod. Thoughts @gkasprow ?

Here's the same measurement on ARTIQ, all channels get sine wave, phase difference is due to gateware. Same situation.

[image]

sbourdeauducq, 2020-01-21:
> Here's the same measurement on ARTIQ (but 50 MHz instead of 240 MHz)

If you upgrade Vivado, you should be able to get 240MHz out of ARTIQ using SAWG. I was able to get SAWG sine output on my board.

marmeladapk, 2020-01-22:
Yellow: Channel0_N Blue: Channel0_P Pink: Channel1_N Green: Channel1_P, measured at TR1 pads (6 and 4)
[image]

gkasprow, 2020-01-22:
I looks loke asymetrically driven balun. Check if there is connection between dac and balun on both p and n wires.

Pozdrawiam,
G.Kasprowicz

[obraz]<https://user-images.githubusercontent.com/24512604/72892696-741fba80-3d17-11ea-8619-1d0d424dac81.png>

marmeladapk, 2020-01-22:
Channel 0 (averaging from 32 triggers) measured at capacitor pads (so directly from on DAC output with C37 and C46 removed)
[image]

gkasprow, 2020-01-22:
How it compares with other channels?

Pozdrawiam,
G.Kasprowicz

marmeladapk, 2020-01-22:
This is asymetrically driven balun: only _N (yellow) is driven and _P (blue) is measured:
[image]

gkasprow, 2020-01-22:
this would explain the issue. maybe connector pad assignment is wrong?

marmeladapk, 2020-01-22:
@gkasprow In the previous screenshot I left one of capacitors unpopulated (on _P line).

marmeladapk, 2020-01-22:
I replaced C46 and C37, still _P line is weaker.

marmeladapk, 2020-01-22:
I removed R41 - jumper after balun on channel 0 so channel 0 is not loaded.
yellow blue - ch0; pink green - ch1 (not modified, for reference)
[image]

So the effect persists.

marmeladapk, 2020-01-22:
Thought for later: I can't see anything special about layer of channel 0. Since it's skewed on all boards that we tried perhaps PCB has some defect?

The effect seems to be caused by balun (currently its output is disconnected and signal from DAC looked fine when balun was disconnected). It's not caused by C37 and C46, I replaced them. Removing C44 didn't help. Maybe pads under balun have some resistive connection to something?

gkasprow, 2020-01-22:
We can check it with VNA tomorrow.

marmeladapk, 2020-01-28:
After amplifier:

[image]

marmeladapk, 2020-01-28:
Swapping amplifiers didn't change the situation, chan 0 is still distorted.

gkasprow, 2020-01-28:
But surprisingly, the signal from balun, routed to the SMA with a piece of the cable looks OK, it has the same amplitude on both channels.