sinara-hw/BaseMod #31: BaseMod Photos

jbqubit, 2020-03-31:
For accounting purposes please post high resolution photo of stuffed BaseMod front and back sides. Also post photo showing stuffed PCBs (one photo of array of boards is fine). 

> BaseMod v2.0: print 20 PCBs, stuff 12

akaminska, 2020-04-07:
1907180PW0007A0108
1907180PW0007A0105
1907180PW0007A0109
1907180PW0007A0115
1907180PW0007A0113
1907180PW0007A0106
1907180PW0007A0110
1907180PW0007A0104
1907180PW0007A0107
1907180PW0007A0114
1907180PW0007A0111
1907180PW0007A0112
[image]
[image]

akaminska, 2020-04-07:
Unpopulated PCBs
[image]
[image]

closed 2020-04-07 by jbqubit